Yveaux/Arduino_Vcc #7: Read Vcc pulls 200uA with LowPower sleep

skjolddesign, 2020-02-15:
Your example VccSleep.ino, makes arduino pull 200 micro Amps in LowPower.
Commenting out Vcc.h code, gives correct LowPower of 4.5 micro Amps.
Tested on Pro micro. 
It's not only your Vcc code that does this, but all readVcc code I found online does this.
I would be glad to see any solution to this.

4.5uA code:
`
#include <Vcc.h>
#include <LowPower.h>

const float VccMin        = 2.0*0.6;  // Minimum expected Vcc level, in Volts. Example for 2xAA Alkaline.
const float VccMax        = 2.0*1.5;  // Maximum expected Vcc level, in Volts. Example for 2xAA Alkaline.
const float VccCorrection = 1.0/1.0;  // Measured Vcc by multimeter divided by reported Vcc

Vcc vcc(VccCorrection);

void setup()
{
  Serial.begin(9600);
}

void loop()
{  
//  float v = vcc.Read_Volts();
//  Serial.print("VCC = ");
//  Serial.print(v);
//  Serial.println(" Volts");
//
//  float p = vcc.Read_Perc(VccMin, VccMax);
//  Serial.print("VCC = ");
//  Serial.print(p);
//  Serial.println(" %");
//
//  delay(200); //delay to allow serial to fully print before sleep

  LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
}
`

skjolddesign, 2020-02-16:
I managed to fix the issue adding ADMUX = _BV(REFS1)
new readVcc code:
```
long readVccFixedLowPower() {
  long result;
  // Read 1.1V reference against AVcc
  ADMUX = _BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1);
  delay(1); // Wait for Vref to settle
  ADCSRA |= _BV(ADSC); // Convert
  while (bit_is_set(ADCSRA,ADSC));
  result = ADCL;
  result |= ADCH<<8;
  result = ( 1023L * 1100L) / result; // Back-calculate AVcc in mV. 1100 is volt on vRef.
  //to fix lowpower drain, ADMUX is changed again.
  ADMUX = _BV(REFS1); //fixes readVcc current drain in lowPower
  Serial.print("Vcc = " );
  Serial.println(result); 
  return result;
}
```

Yveaux, 2020-02-16:
Can you prepare a PR?

danielrheinbay, 2020-06-02:
> Can you prepare a PR?

See #8 :)