Yveaux/Arduino_Vcc #10: If using Minicore version 2.0.8, the compiler produces errors in this library

ericvb, 2020-12-09:
When trying to compile a sketch using this library and the Minicore board manager version 2.0.8 for a 328P 1MHz board, the compiler gives following errors:

```
"C:\\Users\\Eric\\AppData\\Local\\Temp\\arduino_build_851041\\libraries\\VccBatteryPercent\\Vcc.cpp.o"
In file included from C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.h:27:0,
                 from C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:24:
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp: In member function 'float Vcc::Read_Perc(float, float, boolean)':
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:133:22: error: 'l' was not declared in this scope
   typeof (low) _l = (l);           \
                      ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:133:22: note: suggested alternative: '_l'
   typeof (low) _l = (l);           \
                      ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:134:23: error: 'h' was not declared in this scope
   typeof (high) _h = (h);          \
                       ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:134:23: note: suggested alternative: '_h'
   typeof (high) _h = (h);          \
                       ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~
C:\Users\Eric\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\2.0.8\cores\MCUdude_corefiles/Arduino.h:135:35: error: expected ';' before '}' token
   _x < _l ? _l : _x > _h ? _h :_x })
                                   ^
C:\Users\Eric\Documents\Arduino\libraries\VccBatteryPercent\Vcc.cpp:75:12: note: in expansion of macro 'constrain'
     perc = constrain(perc, 0.0, 100.0);
            ^~~~~~~~~

```

When doing a compile of the same sketch using the Minicore version 2.0.7, no compiler errors.

_https://github.com/MCUdude/MiniCore/releases/tag/v2.0.8_
_Changelog:_
_- Fix export compiled binary for Windows computers_
_- Improve "unsafe" macros_
_- This includes abs(), min(), max(), sq(), round() and constrain()_

The problem probably lies in the improved 'unsafe' macros like constrain()...

Yveaux, 2020-12-09:
This library works with regular arduino IDE versions. Furthermore a change of the minicore library apparently broke the constrain function. 
I would suggest to contact the author of minicore.

closed 2020-12-09 by Yveaux

ericvb, 2020-12-09:
For the records: if someone comes searching here.
See https://github.com/MCUdude/MiniCore/issues/164 : it will be corrected in a next version